zamtmn/zcad #7: MTEXT. There is no visualization of the internal molding of MTEXT

veb86, 2018-06-20:
HELLO WORLD

[image]

paolo-caroni, 2020-04-28:
useful documentation of MTEXT format codes can be found on:

https://adndevblog.typepad.com/autocad/2017/09/dissecting-mtext-format-codes.html

https://entercad.ru/acad_aug.en/ws1a9193826455f5ffa23ce210c4a30acaf-63b9.htm

zamtmn, 2020-04-29:
>>https://adndevblog.typepad.com/autocad/2017/09/dissecting-mtext-format-codes.html
>>https://entercad.ru/acad_aug.en/ws1a9193826455f5ffa23ce210c4a30acaf-63b9.htm
Thanks!

zamtmn, 2020-04-29:
It will be difficult to do everything completely, but I will implement the main points

zamtmn, 2025-05-02:
Time passes quickly, but work is slow unfortunately((
Markup support has appeared, but so far it is only loaded and displayed from DXF, it does not affect the appearance of the text.

zamtmn, 2025-05-02:
in autocad:

[image]

in zcad:

[image]