zamtmn/zcad #8: MTEXT. Inside the dxf file, there is no special character protection (\)

veb86, 2018-06-20:
Specific characters of type (\, {,} ...) should have protection (\)

[image]
[image]
[image]
[image]

zamtmn, 2018-06-29:
Fixed in 02ae098

veb86, 2018-10-26:
Almost everything began to work as it should, but there is one problem

ZCAD:
[image]

BRICSCAD:
[image]

The problem is when the following text is entered "\\P"

veb86, 2018-10-26:
Template recorded correctly
Content at the end should be as in the template "\\\\P"

veb86, 2019-01-24:
Add bug file

veb86, 2019-01-24:
[bug.zip](https://github.com/zamtmn/zcad/files/2792447/bug.zip)

zamtmn, 2019-01-29:
[image]
bug.zip mtext content fixed in  2a6292f

commit 02ae0986 (2018-06-29): Fix #8
commit 2a6292fb (2019-01-29): Fix #8
closed 2019-01-29 by zamtmn