zamtmn/zcad #16: Print tool

paolo-caroni, 2020-02-11:
I think that can be cool add a tool for print similar to AutoCAD.
maybe that tool should use .ctb and .pc3 file.

Cheers,
Paolo

zamtmn, 2020-02-17:
Yes, printing subsystem need improvements. *.ctb and *.pc3 fileformats is described somewhere?

paolo-caroni, 2020-02-18:
*.ctb, *.pc3 and *.pmp are proprietary and undocumented file formats.
However they are a simple .txt file compressed in ZIP.
https://github.com/avbrakel/unzip_pc3 (this is a simple tool for decompress pc3, pmp, ctb).

ctb=Color Dependent Plot Style Table
pc3=plotter configuration files
pmp=Custom paper sizes (used/linked by .pc3)

zamtmn, 2020-02-20:
>>However they are a simple .txt file compressed in ZIP.
thanks!

paolo-caroni, 2021-05-08:
If you are interested I have write a PoC for read and write PIA files, the program is honestly orrible, but I have write a specifications of. ctb and .stb that can be interesting.
https://github.com/paolo-caroni/LibrePIA