zamtmn/zcad #26: Cannot render text rntity

zamtmn, 2020-12-01:
See  example file. Looks empty, but contain one text entity
[errorwithtext.zip](https://github.com/zamtmn/zcad/files/5620237/errorwithtext.zip)

Psycho99, 2021-04-05:
I renders pretty nicely on my end.

except for helper box (called system geometry ) that doesn't adjust to get ( around ) rendered text wich make the Horizontal bar of T confused in the box en the last letter a little too close for confort
it's just an adjusting variable to add for capitals in the algorythm that adjust its size .

zamtmn, 2021-04-05:
Stable or git version? please provide screenshot

Psycho99, 2021-04-05:
My bad, I'm on 9.8.5

zamtmn, 2021-04-05:
I have partially correct this error, but not everything depends on me: https://bugs.freepascal.org/view.php?id=38426

commit 3262e162 (2024-02-19): Fix #26, #27.  Реализация ttf на FreeType закончена
closed 2024-02-19 by zamtmn