zamtmn/zcad #36: Save dxf at simplecad not working

ntermaris, 2021-08-10:
The save dxf at simplecad not saving dxf . it saves an empty file. 
I started studing the code of zcad and i want to start a module for surveying inside it. I will try to understand the logic of the electro module to make it similar.
I am a programmer from greece. We had a discussion some years ago. Now i can afford some time to help the project. 
Alex from Greece

ntermaris, 2021-08-10:
After some investigation i found that in module uzeffdxf at function savedxf2000 line 1696 

templatefile.InitFromFile(ProgramPath + 'components/empty.dxf');   

The ProgramPath variable has the issue. 
I run the program from path **C:\Install\zcad-github\source\zcad\cad\bin\x86_64-win64** and the **ProgramPath**  value is 
**C:\Install\zcad-github\source\zcad\cad\** . This means that all the paths are 2 levels up relatavly. This is good for the zcad application but it is not good for other applications using zcad platform. 
If possible to change the save2000dxf to point at app path?

zamtmn, 2021-08-10:
Hi!
Fixed in 9262e0f
Now template file path passed as param to savedxf2000 procedure

closed 2021-08-10 by zamtmn

ntermaris, 2021-08-10:
Thanks Amdrey.