zamtmn/zcad #39: Incorrect file name

zvezdochiot, 2021-10-31:
```sh
uname -a
Linux devuan-zvezdochiot 5.10.0-9-amd64 #1 SMP Debian 5.10.70-1 (2021-09-30) x86_64 GNU/Linux
```
:bug: Path `zcad` (git).
```sh
lazbuild --add-package cad_source/other/laz.virtualtreeview_package/laz.virtualtreeview_package.lpk
Error: (lazbuild) Имя пакета "laz.virtualtreeview_package"
файла "/home/zvezdochiot/WORK/git/zcad/cad_source/other/laz.virtualtreeview_package/laz.virtualtreeview_package.lpk" некорректно.
```

zamtmn, 2021-10-31:
а руками в лазаре этот пакет устанавливается?

zvezdochiot, 2021-10-31:
@zamtmn say:
>а руками

Уточни. При открытии zcad.lpi сразу ругань на неоткрытие laz.virtualtreeview_package. Хотя и директория и файлы есть. Как то ещё надо попробовать?

zamtmn, 2021-10-31:
Запускаешь lazarus в меню Package\Open Package file (*.lpk) открываешь laz.virtualtreeview_package.lpk в окне щелкаешь установить. потом пересобираешь лазарус.
Переделать можно, но пакет не мой, название не я придумывал. у меня в убунте все работало

zvezdochiot, 2021-10-31:
Чего то я совсем в этом Lazarus недопонимаю.
```pas
  cp:=StringCodePage(result);
```
```error
uzeparser.pas(895,7) Error: Can't determine which overloaded function to call
```
Нда.

zamtmn, 2021-10-31:
Похоже у тебя старенький компилятор. fpc что говорит?

zvezdochiot, 2021-10-31:
Да пёс с ним. Поковырялся я с этим Lazarus и вижу что не моё. Не судьба! Будут ещё готовые сборки - потестим, почему бы и нет. Но без возможности собрать самому никакого участия не предусматривается.

closed 2021-10-31 by zvezdochiot

zamtmn, 2021-10-31:
я давно хочу осесть на стабильных версиях laz\fpc, но не очень получается - постоянно всплывают какиенибудь баги-недоделки и приходится юзать транк