zamtmn/zcad #44: Модернизация навигатора или создания чего то нового

veb86, 2022-03-11:
Навигатор который позволит отображать модель здания/ модель подключения и тд
Задача:
- изменить навигатор так что бы глубина его могла регулировать настройками, как сейчас это реализовано для столбцов. Позволит создать иерархию здания, и легкую навигацию внутри нее. Привязанность навигатора должна быть к отдельным пунктам внутри примитивов, что бы внесение шаблонов в этот пункт примитива (@@[GC_HDShortName]@@[NMO_BaseName]@@[GC_HDGroup].@@[GC_NumberInGroup]) вызвало бы правильное расположение примитивов в навигаторе;
- возможность изменять пункты примитива внутри навигатора. То есть, если внутри навигатора выведена мощность, которая указана для примитива индивидуально, это поможет изменить ее прямо в навигаторе
- профили настроек для навигатора. Загрузил профиль: схема подключений и навигатор собирает нужные нам пункты по другому принципу построения иерархии. Кольцо рвем на последнем элементе и указываем что это кольцо. Указав профиль щиты, получим все элементы которые находятся внутри щитов.

Дополняемое и редактируемое

zamtmn, 2022-03-13:
1 пункт уже реализован.
2 - с учетом того что переменные могут быть напрямую не редактируемыми, а например "составлены" по шаблону - нужно думать как сделать
3 - да, это надо

zamtmn, 2022-03-23:
в [0d72dbd](https://github.com/zamtmn/zcad/commit/0d72dbdf4c9da51ce1c8b9b62471e59f1c893a40) добавлено сохранение/загрузка параметров. Наверно стоит еще приделать быстрое переключение по заранее подготовленным параметрам в меню

zamtmn, 2022-03-27:
>>Наверно стоит еще приделать быстрое переключение по заранее подготовленным параметрам в меню
это тоже сделал
Редактирование в навигаторе стоит выделить отдельным репортом имхо. этот закрываю

closed 2022-03-27 by zamtmn