zamtmn/zcad #45: DWG Export

amirj0914, 2022-04-10:
It gives error while exporting DWG.
"Unknown file format"

zamtmn, 2022-04-10:
I don't set such a task, it's too difficult. I tried to import dwg2000, but there are too many unclear points

zamtmn, 2022-07-17:
I tried LibreDWG and changed my mind. This can be done

veb86, 2026-05-26:
> Nice approach. I use also ODA converter but i want to have and a native reader writer in case the user cant install oda converter. Also libredwg check the license if you plan to have it inside the distrubution. I use an info box informing the use to download or ODA converter or LIBREDWG to have dwg support.

I've added DWG support to the master if you'd like to try it out:
https://ci.appveyor.com/project/zamtmn/zcad/build/artifacts