zamtmn/zcad #71: Примитив "Кабель". Неправильное поведение при подключении к устройству

veb86, 2023-09-15:
Примитив "Кабель". Неправильное поведение при подключении к устройству

veb86, 2023-09-15:
[GH0071CableNotConnect.zip](https://github.com/zamtmn/zcad/files/12617373/GH0071CableNotConnect.zip)

veb86, 2023-09-15:
Повторить. ручное построение кабельной линии.
Если кабель сразу попадет на устройство то все норм, но если он с начало встанет на пересечение суперлиний, и потом пойдет на подключение, то получаем такой баг

[image]

zamtmn, 2023-09-19:
Проблема в том что в устройствах стали портится окружности в динамической части при сохранении-загрузке. в данном устройстве в контуре подрезки есть окружность, после сохранения и при загрузке она улетает в ебеня. кабель находит с ней пересечение и считает его оптимальным.

Скорее всего баг добавлен в https://github.com/zamtmn/zcad/commit/857a842a50c766d1db1a142b734b7fa6e8ed7850

Пока временное решение

zamtmn, 2023-09-19:
если включить все слои, то на другом конце (который неверный) кабеля какраз эта окружность))