zamtmn/zcad #78: Не корректное отображение имени стояком

vlsx, 2023-10-22:
На чертеже две пары стояков с разными базовыми менами. У перовой пары "A", у второй "R". В инспекторе базовые и полные имена отображаются верно. Но в фильтре стояков базовые имена отображаются не верно. Из за этого прокладка кабелей методом суперлиний работа не корректно
[Без имени001.zip](https://github.com/zamtmn/zcad/files/13062238/001.zip)
[image]

zamtmn, 2023-10-28:
[image]

Данные стояки имеют как старые переменные имени `RiserName` так и новые `NMO_Name`. на выноске отображена `NMO_Name`, а в инспекторе используется `RiserName`:
```
SetColumnsCount(3,0);
SetColumnParams(0,'Tree','@@[RiserName]','tmpGUIParamSave_NavRis_C0',1);
SetColumnParams(1,'Elevation','@@[Elevation]','tmpGUIParamSave_NavRis_C1',1);
SetColumnParams(2,'Text','@@[Text]','tmpGUIParamSave_NavRis_C2',1)
```

zamtmn, 2023-10-28:
я могу выпилить `RiserName` из инспектора, тебе предлагаю вычистить `RiserName` из своих блоков

zamtmn, 2023-10-28:
Убрал RiserName из настроек инспектора по умолчанию. нужно пересобрать окружение

commit 97a2a17d (2023-10-28): Fix #78
closed 2023-10-28 by zamtmn