zamtmn/zcad #79: Тяжеловесный элемент

vlsx, 2023-11-16:
Очень нужный элемент в схема [но](url) сильно утяжеляет файл. Возможно ли проанализировать причину этого явления?
[DEVICE_EL_VL_SCEMA5_2.dxf.zip](https://github.com/zamtmn/zcad/files/13375974/DEVICE_EL_VL_SCEMA5_2.dxf.zip)

Alligator-1, 2024-03-19:
Проблема находится в слое с именем "Defpoints" если его удалить, то размер кардинально уменьшится
Видимых глазу объектов на этом слое нету, а как посмотреть что там находится я не знаю, не специалист по CAD'ам

Вот уменьшенный вариант вашего файла:
[__DEVICE_EL_VL_SCEMA5_2.zip](https://github.com/zamtmn/zcad/files/14649691/__DEVICE_EL_VL_SCEMA5_2.zip)

Alligator-1, 2024-03-19:
Есть даже более простое решение:
1. Открываем исходный тяжеловесный файл
2. Выделяем ваш этот нужный элемент
3. Копируем его на новый пустой чертеж
4. Готово - файл стал легким