zamtmn/zcad #102: ZCAD падает при закрытии чертежа при определенных условиях

Alligator-1, 2024-03-19:
1. Запустить ZCAD и не двигать и не изменять размер его окна!!!
2. Открыть приложенный чертеж и выделить в нём текст
3. Открыть копию этого чертежа(тут не важно, можно и другой чертеж открыть)
4. Закрыть последний открытый чертеж
5. Падение

[image]
[dxf2000_test.zip](https://github.com/zamtmn/zcad/files/14647176/dxf2000_test.zip)

Alligator-1, 2024-03-19:
[patch102.zip](https://github.com/zamtmn/zcad/files/14657600/patch102.zip)

Этот патч решает падение, но появляется новая проблема в том, что в чекбоксе Dim Style постоянно почти показывает Different

Я не знаю как правильно там логически обработать, может тебе это подскажет куда копать дальше...

Alligator-1, 2024-03-20:
Исправлено "костылём"
https://github.com/zamtmn/zcad/commit/e0907bebb369f9835fa9cd41dae5fe2a58982302

Alligator-1, 2024-03-20:
Возможно там же следует обнулять и остальные члены IVars....🤔

zamtmn, 2024-03-20:
похоже тут надо по другому

zamtmn, 2024-03-20:
Первый патч почти правильный, у меня там опечатка. в случае отсутствия у текущих выдраных примитивов в ivars.CDimStyle надо присвоить текущий размерный стиль. я скопипастил и непоправил. присвоил размерный стиль в текстовый стиль

closed 2024-03-20 by zamtmn