zamtmn/zcad #112: Прямоугольник выделения мышью теряет первоначальный вид после закрытия и открытия чертежа

Alligator-1, 2024-03-20:
1. Отрыть приложенный чертеж
2. Повыделять пустое место - прямоугольник красивый синий/зеленый, полупрозрачный, с диагоналями
3. Закрыть чертеж
4. Открыть заново
5. Повыделять пустое место - прямоугольник залит черным, и видны лишь две грани

(на видео цвет нормального прямоугольника не виден, но можно видеть что отрисовываются все его границы и диагонали)
[image]
[dxf2000_test.zip](https://github.com/zamtmn/zcad/files/14662156/dxf2000_test.zip)

zamtmn, 2024-03-20:
это загадка. сколько не пытался понять почему, так и не понял))

Alligator-1, 2024-03-20:
> это загадка. сколько не пытался понять почему, так и не понял))

Если не трудно - приложи ссылки на код, где отрисовка этого прямоугольника производится

zamtmn, 2024-03-20:
https://github.com/zamtmn/zcad/blob/f1f19e06b0f6aed1902ff5ba7ba3a87c076d399f/cad_source/zengine/zgl/drawers/uzglviewareageneral.pas#L425

Alligator-1, 2024-03-21:
Исправление https://github.com/zamtmn/zcad/commit/8fd5bf4f83215dff9bf73f50ac115df50259ede5

closed 2024-03-21 by Alligator-1