zamtmn/zcad #129: ZCAD падает при старте (0.9.13.0-0.9.12.0)

gtfox, 2024-09-06:
ZCAD падает при старте (0.9.13.0-0.9.12.0)
0.9.11.0 стартует нормально 32 и 64
Может у меня на нубуке нет какого-нибудь нетфреймворка...

Это так, для информации... на домашнем компуктере у меня всё работает...

[image]
[zcadelectrotech_v0.9.13.0_win64_zcad_windows.log](https://github.com/user-attachments/files/16913061/zcadelectrotech_v0.9.13.0_win64_zcad_windows.log)
[zcad_electrotech_v0.9.11.0_win32_zcad_windows.log](https://github.com/user-attachments/files/16913062/zcad_electrotech_v0.9.11.0_win32_zcad_windows.log)

zamtmn, 2024-09-06:
молча падает или создает в %temp% файл zcadcrashreport.txt? его бы тоже приложить

gtfox, 2024-09-06:
[zcadcrashreport.txt](https://github.com/user-attachments/files/16913169/zcadcrashreport.txt)

gtfox, 2024-09-06:
[image]

zamtmn, 2024-09-06:
можно скрин что рядом с ексешником в папке лежит?

gtfox, 2024-09-06:
[image]
[image]

zamtmn, 2024-09-06:
По какойто причине не может загрузить дллку, это релиз скаченый с гитхаба?
простое решение
[image]
заменить тут false на true, зкад перестанет пытаться юзать внешний фритайп и перейдет на встроенную реализацию парсера ттф

gtfox, 2024-09-06:
ага, понял, спасибо!
у меня подозрение на шрифт... может он у меня битый какой-нибудь :)
[Desktop.zip](https://github.com/user-attachments/files/16913267/Desktop.zip)

zamtmn, 2024-09-06:
Нет дело не в шрифте, ругань в момент загрузки длл, до шрифта еще дело не дошло

gtfox, 2024-09-06:
Заработало! Спасибо! :)

Alligator-1, 2024-09-07:
ну может кстати и не хватает VCRuntime какого-то
Не откуда эти DLL включаются в дистрибутив, но к примеру в этом источнике:
https://github.com/ubawurinna/freetype-windows-binaries/releases

компилируют с помощью MSVC и пишут, что для работы библиотеки нужно:

`The freetype libraries use the Universal CRT and therefore requires Visual C++ 2015-2022 Redistributable to be present on system.`

zamtmn, 2024-09-07:
да, надо смотреть чего из зависимостей не хватает

gtfox, 2024-09-07:
Поставил https://aka.ms/vs/17/release/vc_redist.x64.exe
Вернул RD_UseLazFreeTypeImplementation:=False;
Заработало!

gtfox, 2024-09-08:
Было очучение, что проблема ушла... но, как всегда, снизу постучали... :)

Клонировал с гитхаба, собрал, запустил (Visual C++ 2015-2022 Redistributable уже установлен)
[image]
Установка RD_UseLazFreeTypeImplementation:=True; помогает
[zcad_windows.log](https://github.com/user-attachments/files/16921816/zcad_windows.log)
[zcadcrashreport.txt](https://github.com/user-attachments/files/16921818/zcadcrashreport.txt)

zamtmn, 2024-09-30:
Попробуй пожалуйста свежую сборку, там обходной путь, падать больше не должно

gtfox, 2024-10-01:
> Попробуй пожалуйста свежую сборку, там обходной путь, падать больше не должно

Проверил ZCADElectrotech v0.9.14.0-51-ddd75adbc016aeb7419394767b9b06ee9f207223
Не падает при старте

zamtmn, 2024-10-01:
Хоть это и обход а не решение проблемы, всеравно закрою. Проблема в данном случае не в зкаде, а в не разрешенных зависимостях freetype.dll

closed 2024-10-01 by zamtmn