zamtmn/zcad #131: Не устанавливаются пакеты

gtfox, 2024-09-08:
Проект собирается
[image]
При попытке открыть любую форму просит поставить пакеты
[image]
Установка прерывается ошибкой
[image]

Alligator-1, 2024-09-08:
Ооооо, chmreader! Как знакомо )
Я много раз сталкивался с этой проблемой, но каждый раз не помню как её решал

Два варианта, даже три:
1. Удалить исходники ZCAD и попробовать заново
если пользуешься git'ом - то удалить всё из рабочей папки, за исключением .git папки и сделать revert всего чтобы откатиться на вариант из репозитория - попробовать снова сделать все процедуры из установки зкада, которые из командной строки производятся
А если гитом не пользуешься - то удалить исходники и скачать заново
2. Удалить Lazarus и установить заново - и опять же снова команды установки зкада и попытка сборки
3. Ну и третий вариант - всё вместе сразу сделать

Эта происходит, на сколько я могу понять из-за конфликта файлов *.ppu или что-то такое, т.к. в ZCAD были включены исходники из RTL FPC и теперь где-то при каких-то условиях происходит конфликт, по хорошему конечно нужно разобраться и научиться стабильно воспроизводить эту проблему - тогда можно с ней поработать и возможно даже получится решить её раз и навсегда(😁)

Если вдруг заметишь какая последовательность действий приводит к такому - желательно задокументировать и отписаться куда-нибудь

zamtmn, 2024-09-08:
в лазарусе в меню package\Install uninstall packages ...
там удали установленный пакет chmhelppkg

Проблема изза того что зкаду нужен патченый паrет rtl-generics bp из состава fpc. я всё ждал что патчик примут, но надежд всё меньше((

Пока нормального решения нет, только спрятать проблему

gtfox, 2024-09-08:
Всем спасибо! Удаление пакета помогло!

closed 2024-09-08 by zamtmn