zamtmn/zcad #133: Наклон МТекста в стиле не совпадает с Текстом

gtfox, 2024-09-09:
Касаемо текстов:
Создал стиль. Oblique выставил 0,2
[image]
Применил стиль к МТекст - работает!
Применил стиль к Текст - шрифт есть, наклона нет. Вручную выставил наклон 20. Не совпадает с наклоном МТекст-а
Белый - МТекст, Серый - Текст
[image]

А то, что наклон из стиля не применяется к однострочному тексту - это так и задумано?

zamtmn, 2024-09-09:
текст и мтекст работают в автокаде по разному. МТекст берет наклон из стиля, у него нет параметра Oblique.
Текст де имеет параметр Oblique, и хранит свой наклон там. из стиля Oblique заполняется только в момент создания текста.

Связано это с тем что текст более старый примитив и был в автокаде с самых первых версий, когда о стилях еще не заморачивались. Такое поведение тянут как обратную совместимость. МТекст появился емнип в 14 автокаде и его поведение более умное

gtfox, 2024-09-09:
Понял, спасибо! Просто привык в визио, что стили работают на всё. Я вспомнил, что в автокаде последний раз работал до начала САПР-АСУ, а это лет 5 назад :)

closed 2024-09-09 by zamtmn