zamtmn/zcad #134: Сочетания клавиш Ctr+X, Ctrl+D, Ctrl+ЛКМ работают не так, как обычно

gtfox, 2024-09-09:
1. Ctr+X не копирует в буфер = удаление. В офисах эта процедура позволяет вставить вырезанный объект из буфера.
2. Ctrl+D не дублирует выделенное.
3. Режим "Новый набор выбора".
Такой режим по умолчанию в везде (в визио, в офисах, в splane), кроме автокада :)
При этом режиме, при зажатом Ctrl кликаем по объектам и они собираются в набор
Так вот в ZCAD такого не происходит

zamtmn, 2024-09-09:
1 понял
2 должно работать через клипборд? или просто запускать команду COPY?
3 ну мы "копируем" автокад поэтому будем на него похожи)). Надо добавить возможность настройки мышиного поведения

gtfox, 2024-09-09:
2. Визио говорит, что Ctrl+D происходит мимо буфера обмена
На гифке: Ctrl+C->Ctrl+V->Ctrl+D->Ctrl+V
[image]

3. > ну мы "копируем" автокад поэтому будем на него похожи))

Всё так, но Владимир попросил вариант "АнтиАвтокад" в виде единички в инспекторе, и он появился....
Чему я очень рад, т.к. я ненавидел автокад за то, что левая рука вечно лежит на Esc-пе, чтобы снимать не нужные выделения.
Осталось только добавить кнопку Ctrl, которая, пока нажата - отменяет режим "единички" :)
На самом деле это важная штука, какой бы мелкой она ни казалась...

4. Пока писал всё это заметил ещё одну весч:
Копирую устройство, вставляю. Курсор меняется на указание точки вставки, но устройство улетает далеко.
[image]

zamtmn, 2024-09-09:
это не вариант антиавтокад, в автокаде есть точно такойже режим выбора.
4 - этот момент я непомню, в автокаде также или нет - копирование без указания базовой точки считает 0,0,0 базовой, потом вставка без указания базовой точки считает точку под мышкой базовой => полетели))
возможно стоит при копировании брать не 0,0,0 а центр AABB копируемых примитивов. но при вставке всетаки имхо стоит вставлять туда куда смотрит мышь. емнип в офисе он вставляет чуть ниже исходника

gtfox, 2024-09-09:
Да, Ctrl+D в визио вставляет правее и ниже, но я делал привязку к курсору при любой вставке... Это ведь логично и удобно нажать вставку и указать место куда вставлять. 
Я конечно не знаю на сколько это сложно сделать, но по моему для тех, кто чертит эл. схемы не нужна базовая точка в нуле. Примитив должен попасть своей точкой вставки в точку под курсором. Может придумать параметр, который будет это позволять настроить

zamtmn, 2024-09-11:
Сделал чтоб при установленной "1" контрол+клик добавлял примитив в выделение. ты это хотел?
Хотя вот стал писать ответ и подумал что шифт+клик было бы логичнее))

По контрол-Д примитивы дублируются, но я не понял по какому варианту лучше сделать:
1 - вставляются рядом, остаются выделенными = готовы к перетаскиванию мышкой
2 - вставляются под мышку, остаются выделенными = готовы к перетаскиванию мышкой
3 - таскаются за мышкой, вставляются по клику

zamtmn, 2024-09-11:
теперь все вставленные из клипборда примитивы остаются выделенными, для возможного перетаскивания

gtfox, 2024-09-11:
> Сделал чтоб при установленной "1" контрол+клик добавлял примитив в выделение. 

Отлично, спасибо!
Ctrl - принят в винде (при выделении не последовательных файлов в проводнике) и в офисах во всех приложениях
Shift - принят в автокаде. 
В винде Shift+ЛКМ = выделяет группу последовательных файлов.

Что делать: не знаю.
Для обеспечения полной идентичности автокаду - надо shift, для пользователей винды и офиса - привычней Ctrl.
Лучший вариант - дать возможность настройки, либо сделать обе кнопки рабочими - нажав любую из них будет добавляться в набор (хотя логичнее Ctrl - добавляет, Shift - убирает)

> По контрол-Д примитивы дублируются, но я не понял по какому варианту лучше сделать:

по мне так лучше, чтобы после Ctrl+D курсор менялся на указание точки вставки, и поле клика туда вставлялся примитив

Дело в том, что таскание за мышкой - это был костыль из-за ограничений визио, да и splan использовал похожий механизм.
В общем в ZCAD, как мне пока кажется, таскание не нужно.

Когда мы делаем обычное копирование, потом вставку, у нас меняется курсор и отображается то, что мы вставляем, и оно итак уже таскается за курсором, пока мы его не кликнем.
Единственное, что хотелось бы уточнить, что механизм привязки к 0,0 для копирования без базовой точки я бы пересмотрел, хотя бы для электрических схем. 
Хотелось бы при обычном копировании, вставка происходила также, как при вытаскивании из набора девайсов

> теперь все вставленные из клипборда примитивы остаются выделенными, для возможного перетаскивания

На счёт выделения после вставки - я пока не могу сказать надо оно или нет.
Сейчас при доставании из набора элементов выделения не происходит.
В визио оно использовалось только для таскания (если я не ошибаюсь), а здесь оно не надо, поэтому думаю его можно отключить в коде, вдруг понадобится - раскомментировать.

zamtmn, 2024-09-12:
>>Хотелось бы при обычном копировании, чтобы вставка происходила также, как при вытаскивании из набора девайсов
никто не мешает пользователю не один примитив, это не возможно в общем случае.
вообще для втавки я хотел бы сделатб механизм "бегающей" точки вставки, с переключением ее например по ТАБ. но это неблизкие планы - мечты

gtfox, 2024-09-12:
У меня сейчас нет автокада, поэтому нашёл видосик копирования в автокаде
С 0:55 по 1:20
https://www.youtube.com/watch?v=2c23TBaWlV8&t=55s

zamtmn, 2024-09-12:
Я всегда с базовой точкой копировал. никогда даже не задумывался как автокад ее вычисляет при не указании. ок. сделаю так

zamtmn, 2024-09-12:
Чувак в видоск наврал. Автокад берет левую нижнюю точку как точку вставки

gtfox, 2024-09-12:
> ок. сделаю так

Отлично, спасибо!

closed 2024-09-12 by zamtmn