zamtmn/zcad #136: Разлет элементов блока

vlsx, 2024-09-13:
Вновь начали смещаться поля свойств блока при выкидывании блока на чертеж. Данными блоками высоковольтных сетей давно не пользуюсь, но при попытке их использования получается следующее - поля текста свойств  смещаются со своих мест.
[image]

https://github.com/user-attachments/assets/4a60ddeb-a0f0-44b5-96cc-d5d7cebb2bdb

[Разлет элементов.zip](https://github.com/user-attachments/files/16987714/default.zip)

zamtmn, 2024-09-13:
имя блока и в каком файле он определен?

vlsx, 2024-09-13:
EL_VL_QUESTIONNAIRE2_1 
Просто выбрасываю этот блок на новый лист

closed 2024-09-13 by zamtmn