zamtmn/zcad #141: ZCAD падает при закрытии не сохранённого файла

gtfox, 2024-10-01:
[image]
[image]
[zcadcrashreport.txt](https://github.com/user-attachments/files/17215446/zcadcrashreport.txt)
[zcad_windows.log](https://github.com/user-attachments/files/17215448/zcad_windows.log)

Есть еще один момент:
Сохранение происходит даже в уже существующий файл, не предлагая его перезаписать!

zamtmn, 2024-10-01:
Вылет вылечил. Добавил диалог выбора имени файла при сохранении нового чертежа (если файла с таким именем нет) что имеется ввиду под

>Обычное сохранение, без закрытия, происходит даже в уже существующий файл, не предлагая его перезаписать!

ведь подразвмевается что этот файл и был открыт и мы его сохраняем

gtfox, 2024-10-02:
[image]

Нет запроса на перезапись существующего файла

gtfox, 2024-10-02:
> Вылет вылечил.

На простых фигурах - да. А если устройства - вылетает

[image]
[zcadcrashreport.txt](https://github.com/user-attachments/files/17234086/zcadcrashreport.txt)
[zcad_windows.log](https://github.com/user-attachments/files/17234090/zcad_windows.log)

zamtmn, 2024-10-02:
Ты запускаешь зкад по клику на dxf файле на рабочем столе. ты уверен что запускается новая версия? Запускается та, которая ассоциирована с dxf на твоем компе, или файл передается в уже запущенный инстанс зкада. на гифках в заголовке разные версии. при запуске по клитку на диксф этот файл должен открываться и его имя показываться в заголовке. у тебя там "новый файл" я это чинил довольно давно. Чтото идет не так, нужна чистота эксперимента))
С устройствами вылет повторился, разбираюсь. остальное - у меня все ок))

gtfox, 2024-10-03:
>Чтото идет не так, нужна чистота эксперимента))

ZCAD открываю с exe-шника. На обоих гифках при открытии пустой ZCAD и я, каждый раз, нажимаю кнопку "Новый файл". Версии разные потому, что разница по времени 2 часа и я успел скачать ночную  :)  и уже на ней заснял. 

Остался  запроса на перезапись существующего файла

zamtmn, 2024-10-03:
Нашел и поправил еще одну проблему при сохранении, но тут уже не всё так просто(( Правля тут могу сломать чтото другое.
С запросом на сохранение понял(( попробую ченить родить)) 
У тебя есть возможность выкладывать не гифки? а например mp4? гифки тут не останавливаются и не мотаются. тяжело смотреть. можно пропустить нужный момент

zamtmn, 2024-10-03:
Вроде всё поправил

gtfox, 2024-10-03:
> а например mp4?

ага, хорошо

zamtmn, 2024-10-04:
Еще раз))

zamtmn, 2024-11-07:
вроде починено. если что переоткрой

closed 2024-11-07 by zamtmn