zamtmn/zcad #170: Задержки времени при отображении чертежей по сравнению с ранними версиями

vlsx, 2025-02-21:
Снова появились задержки времени при отдалении/приближении чертежей по сравнению с более ранней версией зкада.

[image]

[image]

[Пример.zip](https://github.com/user-attachments/files/18897135/default.zip)

Alligator-1, 2025-02-24:
Мне мало о чем говорит, но может Андрею поможет

[image]

Alligator-1, 2025-02-24:
Да, это для одной из последних версий... где наблюдаются задержки
(смотря время в столбцах нужно учитывать, что там большую задержку ещё профайлер сам по себе вносит)

А версия, где ничего не тормозит - я не нашёл эту версию по хэшу гита (со скриншота в заголовке)

zamtmn, 2025-03-03:
@vlsx попробуй следующее:
[image]

vlsx, 2025-03-03:
Да, помогает. Время рендера снижается с 200/160 до 40/30. Визуально задержек прорисовки чертежа при приближении/удалении  после этого не ощущаю. На тестовой версии программы время рендера 10, но по ощущениям разницы не видно. После перегрузки программы возвращается в первоначальное состояние.

zamtmn, 2025-03-04:
>>После перегрузки программы возвращается в первоначальное состояние.
надо сохранить настройки

Данная проблема (как и предыдущее замедление) - следствие того что я убрал параметр видимости из примитивов. пока не решил что с этим делать, может стоит вернуть. пусть баг открытым поболтается

zamtmn, 2025-03-08:
починено пока так https://github.com/zamtmn/zcad/commit/0295de27853beecfb39e97eab4e8f17077af0253