zamtmn/zcad #175: Самораскрытие списка при выборе дивайсов в инспекторе

vlsx, 2025-03-31:
Постоянно приходится менять свойства группам дивайсов, выбранным в инспекторе. Но при этом автоматически происходит саморакрытие группы, что очень не удобно. Так как чтоб добраться до следующей группы, нужно либо сжать группу либо листать в конец списка. Возможно ли отменить самораскрытие группы? Или если в этом есть практический смысл, то поставить условие выбора - раскрывать группу или нет.

[image]

zamtmn, 2025-03-31:
Эта хренька с раскрытием делалась для того чтобы навигатор оставался в похожем состоянии когда тосуешь последовательность построения дерева тут

[image]
но всё идет не так как задумано(( проверю

zamtmn, 2025-05-22:
улучшил ситуацию с данной проблемой несколькими коментами до https://github.com/zamtmn/zcad/commit/dee2a31174a0390c709b4cf537b227e1bf03d2ea включительно. пжст проверь, закрой если устроит. или дай ситуацию когда самораскрытие всетаки происходит

vlsx, 2025-05-28:
[Запись_2025_05_28_13_06_34_562.zip](https://github.com/user-attachments/files/20473666/_2025_05_28_13_06_34_562.zip)

closed 2025-06-03 by zamtmn